amsbr/OctoPrint-Stats #3: Throws an AttributeError on startup

pelrun, 2016-01-21:
I'm seeing this in the logs every time octoprint starts up; looks like the event handler is being called before on_after_startup does?

```
2016-01-21 09:41:18,710 - octoprint.plugins.stats - INFO - Printer Stats
2016-01-21 09:41:19,206 - octoprint.plugin - ERROR - Error while calling plugin stats
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/plugin/__init__.py", line 204,
in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 515, in on_event
    self.statDB.execute(sql)
AttributeError: 'StatsPlugin' object has no attribute 'statDB'
```

amsbr, 2016-01-24:
Hi, your Octoprint log folder has write permission?

pelrun, 2016-01-24:
Yes, there's nothing odd about the permissions that I can see. The plugin does actually work, too, it just throws this error once per startup.

pelrun, 2016-01-24:
Ok, I've found the culprit. Octoprint raises a CONNECTED event before it actually finishes running all the on_after_startup callbacks. (I've got auto-connect on startup configured, obviously).

The fix is pretty simple, change on_after_startup(self) to on_startup(self,host,port) instead. No events occur before that point, but everything else needed is available (e.g. the path to the octoprint directory.)

amsbr, 2016-01-24:
Thank you! I apply the fix. Anyway I will soon release a new version with improvements.

amsbr, 2019-09-01:
Hi @pelrun!

Please try the renewed version, 2.0.0

closed 2019-09-01 by amsbr