amsbr/OctoPrint-Stats #8: Months showing wrong events.

arose62, 2017-08-01:
I have done only 1 print so far this month, but the graphs are showing large numbers of all events for the current month.
Below are screenshots - the History image shows correctly only 1 job in August.
[image]
This screenshot shows how Print-Stats has lots and lots of events for August (which is incorrect).
[image]

gztproject, 2018-01-11:
I have the same problem (OctoPrint 1.3.6 running on OctoPi 0.14.0), events from November are showing up in January. It seems the problem is in back-end as the JSON data shows the same as charts.
I attached print history .csv and JSON response in .zip

[image]

[history.zip](https://github.com/amsbr/OctoPrint-Stats/files/1622221/history.zip)

amsbr, 2018-01-23:
Hi,

I will prepare a new release in some days.

gztproject, 2018-09-06:
I just submitted a pull request, so far the testing looks great! :)
[image]

amsbr, 2019-09-01:
Hi @arose62 @gztproject @sajovicd!

Please try the renewed version, 2.0.0

closed 2019-09-01 by amsbr