amsbr/OctoPrint-Stats #9: Octoprint 1.4.0 error

mcecchi, 2017-08-25:
Error during startup:

2017-08-25 14:08:54,529 - octoprint.plugin - ERROR - Error while calling plugin stats
Traceback (most recent call last):
  File "c:\devel\octoprint\src\octoprint\plugin\__init__.py", line 229, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "c:\devel\octoprint-stats\octoprint_stats\__init__.py", line 515, in on_event
    self.statDB.execute(sql)
AttributeError: 'StatsPlugin' object has no attribute 'statDB'

Tnx

Dbroqua, 2017-09-22:
Hello,

I have the same issue with Octoprint on a Raspberry Pi.

amsbr, 2018-01-23:
Hi,

I will prepare a new release in some days.

aguindehi, 2019-03-04:
Did you find time for that release? I could not find it. It would be cool to eliminate that error from my logfiles... Thanks for your plugin!

amsbr, 2019-09-01:
Hi @mcecchi @aguindehi @Dbroqua!

Please try the renewed version, 2.0.0

closed 2019-09-01 by amsbr