amsbr/OctoPrint-Stats #10: Print Stats blank after fresh install

beecroftst, 2017-11-21:
[image]

Now sure what the issue is, installed using Octoprint Plugin manager, with a couple other plugins, restarted Octoprint and the Print stats are blank.

Any ideas?

Thanks,
Steve

amsbr, 2018-01-23:
Hi Steve,

please send details of your OctoPrint version. I will prepare a new release in some days.

Erutan409, 2018-04-01:
I'm having the same issue.  Is there any ETA on an update for this?  Seemed to stop working all of a sudden after a recent upgrade.

amsbr, 2018-04-03:
Hi all,

I building a new version with MongoDB, for correct all bugs and add new features. I intend to release this on some days.

Thank you

Erutan409, 2018-04-03:
I uninstalled a couple plugins.  After doing so, the issue corrected itself.  One of them I recall was the YouTube Live plugin.  Don't know if this affecting other people.  Thought I'd share.

ericmathison007, 2018-09-07:
I'm having the same issue. I installed your plugin and see the same blank screen as the screenshot in the first post.

I'm using OctoPrint 1.3.9.

KenwoodFox, 2018-12-24:
I too am having this issue, Hope this can be fixed soon.

Using OctoPrint 1.3.9

KenwoodFox, 2019-01-28:
Im using the cost plugin, and the above image also uses the cost plugin, maybe i will try uninstalling cost

amsbr, 2019-09-01:
Hi @beecroftst @Erutan409 @KenwoodFox!

please try the renewed version 2.0.0

closed 2019-09-01 by amsbr