amsbr/OctoPrint-Stats #18: Update Error 2.0

Leetspeak1286, 2019-09-01:
Aktualisiere gerade, bitte warten.
++++++++++++++++++++++++++++++++++++
Aktualisiere Printer Stats auf 2.0.0
++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/2.0.0.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/2.0.0.zip
Downloading https://github.com/amsbr/OctoPrint-Stats/archive/2.0.0.zip (1.2MB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.0) (1.3.11)
Collecting tinydb (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/d7/f9/0e871cbf0da678cf1780609dc6aef26a5ed544c86733fc1ceaf134fce52c/tinydb-3.13.0-py2.py3-none-any.whl
Collecting pandas (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-oA1iJk/pandas/setup.py", line 746, in <module>
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 724, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 846, in resolve
dist = best[req.key] = env.best_match(req, ws, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1118, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1130, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 791, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-w_2B3D/numpy-1.17.1/setup.py", line 31, in <module>
RuntimeError: Python version >= 3.5 required.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-oA1iJk/pandas/
Die Aktualisierung wurde nicht erfolgreich abgeschlossen. Bitte konsultiere <code>octoprint.log</code> und <code>plugin_softwareupdate_console.log</code> für Details.

Racailloux, 2019-09-01:
Hello,

I got the same here.
I believe that oprint is using python 2.7 natively and it seems that the updater needs v3.5 or higher. (Both are installed on my R.Pi).

amsbr, 2019-09-01:
Hi!

I go investigate this. I developed this in macOS with Python 2.7.15, and install it on my board OPi Win Plus with Python 2.7.12

@Racailloux please share your log.

Racailloux, 2019-09-02:
@amsbr : Here they are.
[octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3564504/octoprint.log)
[plugin_softwareupdate_console.log](https://github.com/amsbr/OctoPrint-Stats/files/3564505/plugin_softwareupdate_console.log)

nizongdai, 2019-09-02:
I have the same error, it keeps saying "pandas: still running..." in the dialog window.

akurz42, 2019-09-02:
Update went through ok for me - it just took hours constantly showing "pandas: still running...", but in the end it succeeded.

Alex

amsbr, 2019-09-02:
Yes,

"pandas: still running..." is inconvenient. It's because pandas use stats libraries (as numpy), to compile this in arm boards I waited a long time, in my board (2gb ram and 4 cores) I waited 15 minutes.

In computer 64bits, minus of 1 minute.

But I go to investigate error reported by @Leetspeak1286 and @Racailloux

akurz42, 2019-09-02:
Yes, I've just corrected my initial statement as well. It didnt take hours on my Raspi either. No idea why I wrote that :)

Alex

mloebl, 2019-09-02:
I've got the same issue as Racailloux: RuntimeError: Python version >= 3.5 required.

GilDev, 2019-09-02:
Got that same `RuntimeError: Python version >= 3.5 required.` error on Raspberry Pi 4 (4 GB) running OctoPi 0.17.0 and OctoPrint 1.3.11.

amsbr, 2019-09-02:
Hi

@Leetspeak1286 @Racailloux @mloebl @GilDev try run this first:

> pip install numpy

in your env. Issue related [#6667](https://github.com/pypa/pip/issues/6667)

nizongdai, 2019-09-03:
> (Edit: This used to say it took hours. That is not correct! It was maybe half an hour total on a Raspi 3.)
> 
> Update went through ok for me - it just took very long, constantly showing "pandas: still running...", but in the end it succeeded.
> 
> Alex

thank you, I wait for 50 mins on my Pi3, and finally its success.

Michel83, 2019-09-03:
Hi,
i think i have the same problem

`Installing collected packages: pandas, OctoPrint-StatsRunning setup.py install for pandas: startedRunning setup.py install for pandas: still running...Running setup.py install for pandas: still running...Running setup.py install for pandas: still running...Fehler!Could not install plugin, unknown error, please consult octoprint.log for details`

`pip install numpy` i have tried and it is installed 

The error log is empty 
`2019-09-03 11:26:40,673 > Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
2019-09-03 11:27:06,620 > Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,622 > Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,623 > Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,625 > Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,626 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,631 > Requirement already satisfied: Flask-Principal<0.5,>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,633 > Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,635 > Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,638 > Requirement already satisfied: werkzeug<0.16,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,639 > Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,640 > Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,642 > Requirement already satisfied: markdown<3.1,>=3.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,643 > Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,645 > Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,647 > Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,648 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,649 > Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,651 > Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,652 > Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,653 > Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,655 > Requirement already satisfied: requests<3,>=2.21.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,656 > Requirement already satisfied: semantic_version<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,658 > Requirement already satisfied: psutil<5.7,>=5.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,660 > Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,662 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,664 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,665 > Requirement already satisfied: chainmap<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,667 > Requirement already satisfied: future<0.18,>=0.17.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,668 > Requirement already satisfied: scandir<1.11,>=1.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Stats==2.0.0)
[... cut ...]
2019-09-03 11:27:06,714 > Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from websocket-client<0.57,>=0.56->OctoPrint->OctoPrint-Stats==2.0.0)
2019-09-03 11:27:06,716 > Installing collected packages: pandas, OctoPrint-Stats
2019-09-03 11:27:06,718 > Running setup.py install for pandas: started
2019-09-03 11:28:16,811 > Running setup.py install for pandas: still running...
2019-09-03 11:29:43,922 > Running setup.py install for pandas: still running...
2019-09-03 11:39:16,267 > Running setup.py install for pandas: still running...
`

swill010101, 2019-09-03:
Same issue here... Thanks for looking into it.

SamPovilus, 2019-09-03:
@amsbr Same issue on an octoprint, tried to update pandas and numpy through pip3 manually.

christianuhlmann, 2019-09-03:
installing nymphy in octoprint enviroment solves the error:
`/home/pi/oprint/bin/pip install numpy`

SamPovilus, 2019-09-03:
@christianuhlmann that was  good hint, that worked for me. Thanks

Racailloux, 2019-09-03:
@amsbr The numpy was installed, but I still get the same updating error (see attached logsà. Strange...

[octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3571051/octoprint.log)
[plugin_softwareupdate_console.log](https://github.com/amsbr/OctoPrint-Stats/files/3571052/plugin_softwareupdate_console.log)

I had to install pip that is not installed and available by apt or apt-get.
I had to follow this recommandations to make it installed : https://medium.com/@isma3il/install-python-3-6-or-3-7-and-pip-on-raspberry-pi-85e657aadb1e
If that can help anyone else.

christianuhlmann, 2019-09-03:
> @amsbr The numpy was installed, but I still get the same updating error (see attached logsà. Strange...
>
> [octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3571051/octoprint.log) [...]

please use the octoprint enviroment there is pip already available
see my comment in comment:

> installing nymphy in octoprint enviroment solves the error:
> `/home/pi/oprint/bin/pip install numpy`

mloebl, 2019-09-03:
I think @christianuhlmann's solution was what worked for me, thank you!

Any way to run the update script again? After 10-15mins my connection via VPN got interrupted, and now no stats and errors in the log :(

Michel83, 2019-09-03:
> 
> 
> I think @christianuhlmann's solution was what worked for me, thank you!
> 
> Any way to run the update script again? After 10-15mins my connection via VPN got interrupted, and now no stats and errors in the log :(

I have the same, at the moment tested

Michel83, 2019-09-03:
`pi@octopiEnder:~ $ /home/pi/oprint/bin/pip install numpy
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in ./oprint/lib/python2.7/site-packages (1.16.2)`

amsbr, 2019-09-03:
@Michel83 try to install all depency manually.

`pip install pandas`
`pip install tinydb`

After plugin install. Install is a long time in arm boards because numpy and pandas is hard for compile.

In OctoPrint interface the log display "still running" for a long time +/- 30min (depends board).

amsbr, 2019-09-03:
> I think @christianuhlmann's solution was what worked for me, thank you!
> 
> Any way to run the update script again? After 10-15mins my connection via VPN got interrupted, and now no stats and errors in the log :(

@mloebl the dependency install run during a long time, but after some time show OctoPrint and Stats, pandas are the best library for this stats (and future releases of the plugin), but need a long time for compile

mloebl, 2019-09-03:
@amsbr That's what worries me, both say installed:
`pi@3dprint01:~/OctoPrint-Stats $ /home/pi/oprint/bin/pip install pandas
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pandas in /home/pi/oprint/lib/python2.7/site-packages (0.24.2)
Requirement already satisfied: python-dateutil>=2.5.0 in /home/pi/oprint/lib/python2.7/site-packages (from pandas) (2.6.1)
Requirement already satisfied: pytz>=2011k in /home/pi/oprint/lib/python2.7/site-packages/pytz-2016.3-py2.7.egg (from pandas) (2016.3)
Requirement already satisfied: numpy>=1.12.0 in /home/pi/oprint/lib/python2.7/site-packages (from pandas) (1.16.5)
Requirement already satisfied: six>=1.5 in /home/pi/oprint/lib/python2.7/site-packages (from python-dateutil>=2.5.0->pandas) (1.8.0)
pi@3dprint01:~/OctoPrint-Stats $ /home/pi/oprint/bin/pip install tinydb
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: tinydb in /home/pi/oprint/lib/python2.7/site-packages (3.13.0)
`

2019-09-03 18:44:43,767 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin stats
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 290, in on_api_get
    self.refreshFull(filter_param=self.current_filter)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 438, in refreshFull
    connected = self.filterEvent("CONNECTED", group = "ym", filterp=filter_param)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 379, in filterEvent
    event_data = self.statDB.getData(self.statDB.query(search_data, 'events'))
AttributeError: 'StatsPlugin' object has no attribute 'statDB'
2019-09-03 18:44:43,775 - tornado.access - ERROR - 500 GET /api/plugin/stats?filter=current_year (192.168.0.99) 17.92ms

amsbr, 2019-09-03:
It's a startup? I observed that a startup show no attribute for 'statDB', but after events (connect, print, etc) the database is generated at log folder.

Try it, if only at startup I going to correct this

Michel83, 2019-09-04:
> 
> 
> @Michel83 try to install all depency manually.
> 
> `pip install pandas`
> `pip install tinydb`
> 
> After plugin install. Install is a long time in arm boards because numpy and pandas is hard for compile.
> 
> In OctoPrint interface the log display "still running" for a long time +/- 30min (depends board).

Hello Thanks for the support, it dosent work
`DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pandas
Requirement already satisfied: pytz>=2011k in ./oprint/lib/python2.7/site-packages (from pandas) (2018.4)
Requirement already satisfied: numpy>=1.12.0 in ./oprint/lib/python2.7/site-packages (from pandas) (1.16.2)
Requirement already satisfied: python-dateutil>=2.5.0 in ./oprint/lib/python2.7/site-packages (from pandas) (2.6.1)
Requirement already satisfied: six>=1.5 in ./oprint/lib/python2.7/site-packages (from python-dateutil>=2.5.0->pandas) (1.11.0)
Installing collected packages: pandas
Successfully installed pandas-0.24.2`

When i now start the Update in Octoprint, at 20 minutes lose the ssh conection and 👍 

[image]

dont4u, 2019-09-04:
I got the same here.
I believe that oprint is using python 2.7 natively and it seems that the updater needs v3.5 or higher.

amsbr, 2019-09-04:
@Michel83 @dont4u please send the logs.

mloebl, 2019-09-04:
@amsbr I get the errors in the logs when I go to the PrinterStats sectionin the UI, not at startup. Also trying to figure out now why on every other reboot, Octoprint is coming up in safe mode.

larp-welt, 2019-09-04:
Same problem for me. I'm very tempted to say: the problem is pandas needing python 3.5. Why do I belive this:

Output from pip:

`RuntimeError: Python version >= 3.5 required.`

Or what they say on there homepage:

> Warning Starting with the 0.25.x series of releases, pandas only supports Python 3.5.3 and higher. See Dropping Python 2.7 for more details.

Source: [https://pandas.pydata.org/pandas-docs/version/0.25/whatsnew/v0.25.0.html](https://pandas.pydata.org/pandas-docs/version/0.25/whatsnew/v0.25.0.html) (July 18, 2019)

I still attach the log from pip to this post.

I'm runnig on:

```
------------------------------------------------------------------------------
OctoPrint version : 1.3.11
OctoPi version    : 0.16.0
------------------------------------------------------------------------------
```

[error.log](https://github.com/amsbr/OctoPrint-Stats/files/3575713/error.log)

Racailloux, 2019-09-05:
Solved for me : Installing using the absolute patu of hte pip from oprint folder.
Waiting 45min (R.Pi 3B) of installation
Loosing all previous stats from v1.0.0

Michel83, 2019-09-05:
> 
> 
> @Michel83 @dont4u please send the logs.

Hello, my problem is solved.

I simply let the printer run overnight with the update, at the next morning power reset and install again the update then is all working.

`pip install pandas
pip install tinydb`

Unfortunately, I can not say exactly if the code has helped or the installer from octoprint

Thanks for the great support

amsbr, 2019-09-05:
> Solved for me : Installing using the absolute patu of hte pip from oprint folder.
> Waiting 45min (R.Pi 3B) of installation
> Loosing all previous stats from v1.0.0

the previous database stays stored at log folder (bak extension), removing bak extension it migrate to the new version.

amsbr, 2019-09-05:
> Hello, my problem is solved.
>
> I simply let the printer run overnight with the update, at the next morning power reset and install again the update then is all working.
>
> `pip install pandas pip [...]

Yes. One step of install the plugin is the install these dependencies. It processes this command, but via the installer, it shows only still running during some time.

FractalEngineer, 2019-09-07:
Hey thanks @christianuhlmann and @amsbr ; I was having the same issue as OP; installed numpy and dependencies through SSH and the plugin installed and runs perfectly

jeracom, 2019-09-08:
same error....

amsbr, 2019-09-08:
> same error....

Hi @jeracom! Do you received message "still running"?

jeracom, 2019-09-08:
No just an error message and the installation failed

amsbr, 2019-09-09:
> No just an error message and the installation failed

Try run command in your venv instance of OctoPrint:

`pip install pandas; pip install tinydb`

jeracom, 2019-09-09:
ok, solved

swill010101, 2019-09-09:
Will a new version be released with a fix or should we move forward with these fixes? Thanks...

amsbr, 2019-09-09:
You can proceed with these procedures.  I am still evaluating how to minimize the impact of pandas installation.

jeracom, 2019-09-09:
It's good. Is nothing for people who are not familiar with the area ...

Greetings, Jens

xenon2008, 2019-09-09:
Since i have updated to Version 2.0 my Octopi on my RPI3b+ is Very slow!
I dont get any Notification if a Timelapse is rendering etc.
Also the Prints stops & sucks for seconds...
If i disable Printerstats everything works find!

It seems the new Version generates high load on the OCtopi?
What sould i do?
I already have tried do reinstall it, but without success,

Best Regards
Mark

planetar, 2019-09-10:
I was trying to do a first install (not an update like the issue title says) through octoprint and it failed reproducibly with errors installing pandas.
I tried the workaround steps mentioned her (/home/pi/oprint/bin/pip install numpy like christianuhlmann recommended, from the command line) Thereafter trying pip install pandas and it still needs pip >= 3.5

so, no install for me. 
Is there a way to get a former version of OP-stats or tell pip to install a pandas version that was still ok with 2.7?

all the best
daniel

b14ckyy, 2019-09-18:
I had the same Update error in the first place. So i tried to reinstall with the newest version and get the same issue now as @planetar 
Unfortunately my database is gone now and all my previous statistics :( Also backup restore did not help

t3g, 2019-09-21:
I have exactly the same issue.
I think the main problem is, that pandas "Dropped Python 2 support".
And this was around the time amsbr created it :/.
See here the official statement of July 18, 2019 for pandas 0.25.0:
https://pandas.pydata.org
0.24.X is the last to support python 2.7. See here in the release notes:
https://pandas.pydata.org/pandas-docs/version/0.24/whatsnew/v0.24.2.html
right now it is 0.25.1 and there the support is dropped:
https://github.com/pandas-dev/pandas/releases

Hope it helps understanding the problem... a solution would be, that the whole octoprint code will be converted to python 3.X...

ZeekWorld, 2019-09-24:
I've had installed 1.0.0 and was working without issues. An update became available 2.0.0, update failed to installed.
I uninstalled the plug in and re installed. Still no luck. See error below.
Running Octoprint (OctoPi) Version 1.3.12rc3

Installing plugin "Printer Statistics" from https://github.com/amsbr/OctoPrint-Stats/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Downloading https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.0) (1.3.12rc3)
Collecting tinydb (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/dc/90/ced9f2a57513eb75dce70760e55967ceedbf2dc158b6a66f1bf87ac8fab0/tinydb-3.14.1-py2.py3-none-any.whl
Collecting pandas (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "", line 1, in
File "/tmp/pip-install-umQrj2/pandas/setup.py", line 746, in
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/init.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/init.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 724, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/init.py", line 846, in resolve
dist = best[req.key] = env.best_match(req, ws, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/init.py", line 1118, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/init.py", line 1130, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 791, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in exit
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in exit
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-nQl7wJ/numpy-1.17.2/setup.py", line 31, in
RuntimeError: Python version >= 3.5 required.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-umQrj2/pandas/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

mplogas, 2019-09-26:
Workaround for now:
downgrade panda to 0.24.2 (this will take some time on your pi)
` /home/pi/oprint/bin/pip install pandas==0.24.2` 

install / upgrade plugin:
```
Installiere Plugin "Printer Statistics" von https://github.com/amsbr/OctoPrint-Stats/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
//...
Installing collected packages: OctoPrint-Stats
Running setup.py install for OctoPrint-Stats: started
Running setup.py install for OctoPrint-Stats: finished with status 'done'
Successfully installed OctoPrint-Stats-2.0.0
Fertig!
```

raulttro, 2019-09-27:
> Workaround for now:
> downgrade panda to 0.24.2 (this will take some time on your pi)
> ` /home/pi/oprint/bin/pip install pandas==0.24.2`
>
> install / upgrade plugin:
>
> ```
> Installiere Plugin "Printer [...]

Tried this but I still get the RuntimeError: Python version >= 3.5 required error. Other suggestions didn't work either. 

I've ran the command from Putty. Did I do something wrong here?

mplogas, 2019-09-29:
That's weird. I ran it from commandline as well. Do you use other plugins that require pandas and may have updated pandas again? 

Can you please check your version of pandas by issueing `pip list | grep pandas` or 

```
import pandas as pd
pd.__version__
```

SargonofAssyria, 2019-10-01:
With the procedure above I finally got v2.0 installed. BUT after the install one of the cores of the cpu stayed at 100% resulting in very bad printing. So I had to revert back to previous version.
Anyone a solution for that?
B.t.w. I had to increase my /tmp to 384 Mb to get in installed.

amsbr, 2019-10-11:
Hi,

Please try the new release, dedicated to reducing CPU usage, and to fix event registration during print. 

[2.0.2](https://github.com/amsbr/OctoPrint-Stats/releases/tag/2.0.2)

swill010101, 2019-10-12:
Still didn't work..... Python 3.5 required....

Have a great day!

Tom

bgiovanny, 2019-12-14:
Installing plugin "Printer Statistics" from https://github.com/amsbr/OctoPrint-Stats/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Downloading https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.2) (1.3.12)
Collecting tinydb
Downloading https://files.pythonhosted.org/packages/9b/83/2d46115b89640e9b85b94df47216547396e94125245dd3ade186036ce976/tinydb-3.15.1-py2.py3-none-any.whl
Collecting pandas
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-n5bIQc/pandas/setup.py'"'"'; __file__='"'"'/tmp/pip-install-n5bIQc/pandas/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-n5bIQc/pandas/pip-egg-info
cwd: /tmp/pip-install-n5bIQc/pandas/
Complete output (51 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-n5bIQc/pandas/setup.py", line 746, in <module>
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 720, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 782, in resolve
replace_conflicting=replace_conflicting
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1065, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1077, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 787, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-Gpmr7K/numpy-1.18.0rc1/setup.py", line 31, in <module>
RuntimeError: Python version >= 3.5 required.
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

swill010101, 2019-12-14:
????

Have a great day!

Tom

bgiovanny, 2019-12-14:
When I install the plugin i get that error...

swill010101, 2019-12-14:
Thanks.....
The email I received looked like it was an answer to the issue.

I've uninstalled this 'previously great' add-in from both of my
printers.....

Have a great day!

Tom

bgiovanny, 2019-12-14:
I still get the same error!
Can you help me solve the problem?
Thanks

swill010101, 2019-12-14:
Me too.....

I've uninstalled on both my printers.

Have a great day!

Tom

[2 more replies]