amsbr/OctoPrint-Stats #20: Installation fail on Buster Octoprint

Key0nee, 2019-09-07:
Hello I just found your awesome plugin and wanted to use it on my newly created octoprint server running on a Pi4 (Buster)....  I tried twice from the plugin manager and got this error.  attached below is my octoprint.log file.... not a coder but willing to help with more details and testing...

Thanks
K

Error!
Could not install plugin, unknown error, please consult octoprint.log for details

[octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3586916/octoprint.log)

amsbr, 2019-09-08:
@Key0nee view more details in #18

amsbr, 2019-09-11:
@Key0nee the instructions in #18 work for you?

amsbr, 2019-09-13:
Very good!

Thank you for feedback

Key0nee, 2019-09-13:
Well... I couldn't leave well enough alone and decided to try and upgrade...   not sure why or even if needed... I just figured its probably better to be running the latest stuff...

after running sudo apt upgrade 

OctoPrint-Stats no longer works it just shows the busy "throbber" as if it is trying to fetch data and never displays anything...   as mentioned above about the "numpy" dependency, it maybe note worthy that the Bed Level Visualizer plugin still works properly after the upgrade

if I can help with details and logs,  just tell me which ones you need..

Thanks
K

amsbr, 2019-10-11:
Hi,

Please try the new release, dedicated to reducing CPU usage, and to fix event registration during print. 

[2.0.2](https://github.com/amsbr/OctoPrint-Stats/releases/tag/2.0.2)