amsbr/OctoPrint-Stats #21: Does not load after 2.0 Update

ngroulx, 2019-09-10:
I was able to "successfully" install the 2.0 update, however, the plugin never loads.  I tried to uninstall the plugin and reinstall it, but that did not make any change.

[image]

amsbr, 2019-09-10:
Hi @ngroulx please send log files generated after click in filter button.

ngroulx, 2019-09-10:
Please find attached my log file.

[octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3597352/octoprint.log)

amsbr, 2019-09-10:
Do you can share with me your stats db file, octoprint_stats.json?

ngroulx, 2019-09-10:
Attached are the requested files. Here are some more details:
1. The stats have not loaded for me for about a week now since I did the upgrade.
2. The upgrade had some issues and failed, but after following one of the other threads I got it to successfully install
3. No stats all week as per the image in my original post.
4. I did some digging this morning prior to this post and found that I had no **octoprint_stats.db**, I only had **octoprint_stats.db.bak**
5. I copied the .bak to .db -> after some time I still had no data.
6. Several hours later (after I sent the last log file), data has suddenly appeared!

I'm not sure if maybe it needed some time to update the old db once I restored the .bak?

Hopefully, this information helps you, or someone else. I would, however, like to know how I can purge the db and start with fresh data?

Thank you
[octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3597740/octoprint.log)
[octoprint_stats.zip](https://github.com/amsbr/OctoPrint-Stats/files/3597721/octoprint_stats.zip)

amsbr, 2019-09-11:
@ngroulx please install this version:

[2.0.1pre](https://github.com/amsbr/OctoPrint-Stats/archive/2.0.1pre.zip), it's fix the issue for NaN value of column ptime.

Please send feedback if this work properly.

ngroulx, 2019-09-11:
I installed 2.0.1pre, after which it required a restart. Octoprint then failed to start properly and then loaded in safe mode. After a couple more restarts, it loaded without safe mode, however, the data is still not displaying.

[image]
[image]
[Octoprint_stats-20190911-1224.zip](https://github.com/amsbr/OctoPrint-Stats/files/3601789/Octoprint_stats-20190911-1224.zip)

scottrini, 2019-09-13:
I installed this for the first time straight to 2.0 and none of the stats load, probably seeing the same issue.  When making the request, the page is throwing a 500 internal server error.  The log file shows::

2019-09-13 12:06:38,237 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin stats
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 290, in on_api_get
    self.refreshFull(filter_param=self.current_filter)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 466, in refreshFull
    print_failed_count = self.filterEvent("PRINT_FAILED", group = "ym", filterp=filter_param)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 392, in filterEvent
    event_df['ptime'] = event_df['ptime'].apply(self.formatNum)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pandas/core/frame.py", line 2927, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pandas/core/indexes/base.py", line 2659, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'ptime'
2019-09-13 12:06:38,246 - tornado.access - ERROR - 500 GET /api/plugin/stats?filter=current_year (::ffff:10.0.106.37) 210.43ms

amsbr, 2019-09-13:
@scottrini please try this [2.0.1pre](https://github.com/amsbr/OctoPrint-Stats/releases/tag/2.0.1pre)

scottrini, 2019-09-13:
@amsbr That did work.  Upgraded on one of mine and it loaded and displayed stats.

scottrini, 2019-09-14:
So I have 3 octoprints running and this worked on one of them, but not on the other two. New error output below.  Only difference I missed is the line 463 vs 466

2019-09-14 08:28:29,471 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin stats
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 290, in on_api_get
    self.refreshFull(filter_param=self.current_filter)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 463, in refreshFull
    print_done_count = self.filterEvent("PRINT_DONE", group = "ym", filterp=filter_param)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 392, in filterEvent
    event_df['ptime'] = event_df['ptime'].apply(self.formatNum)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pandas/core/frame.py", line 2927, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pandas/core/indexes/base.py", line 2659, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item

amsbr, 2019-10-11:
Hi,

Please try the new release, dedicated to reducing CPU usage, and to fix event registration during print. 

[2.0.2](https://github.com/amsbr/OctoPrint-Stats/releases/tag/2.0.2)

ngroulx, 2019-10-11:
I just installed 2.0.2, and right away things seem to be working much better! Thank you! I'll report back should any issues creep up.

condatek, 2019-10-12:
Hello, I have version 2.0.2 installed but similar error appear and I'm not able to see stats like in this post: https://github.com/amsbr/OctoPrint-Stats/issues/21#issuecomment-530458036

2019-10-12 14:28:28,071 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin stats
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 293, in on_api_get
    self.refreshFull(filter_param=self.current_filter)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 469, in refreshFull
    print_failed_count = self.filterEvent("PRINT_FAILED", group = "ym", filterp=filter_param)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 395, in filterEvent
    event_df['ptime'] = event_df['ptime'].apply(self.formatNum)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pandas/core/frame.py", line 2927, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pandas/core/indexes/base.py", line 2659, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'ptime'
2019-10-12 14:28:28,084 - tornado.access - ERROR - 500 GET /api/plugin/stats?filter=current_month (192.168.1.21) 461.11ms

SargonofAssyria, 2019-10-14:
For what it is worth: I get **exactly** the same errors. [ last line: different ip and longer processing time ;) ]

oblivionratula, 2019-10-17:
Installed 2.0, had same problems reported here. Tried 2.0.2, restarted, but still seeing the following:

2019-10-16 21:04:57,065 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin stats
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.11-py2.7.egg/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 293, in on_api_get
    self.refreshFull(filter_param=self.current_filter)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 469, in refreshFull
    print_failed_count = self.filterEvent("PRINT_FAILED", group = "ym", filterp=filter_param)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 395, in filterEvent
    event_df['ptime'] = event_df['ptime'].apply(self.formatNum)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pandas/core/frame.py", line 2927, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pandas/core/indexes/base.py", line 2659, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'ptime'
2019-10-16 21:04:57,086 - tornado.access - ERROR - 500 GET /api/plugin/stats?filter=current_month (192.168.1.6) 385.28ms

sajovicd, 2019-10-22:
I have same problem, running version 2.02

rodolfo61, 2019-10-24:
I have same problem, running version 2.02

trininja2k, 2019-10-29:
I'm stuck with the same issue after a clean OctoPi install. Had to manual turn the wheels for pandas and co to get 2.0.2 installed and afterwards just a spinning wheel on the overview webpage and when trying to filter.

PR #31 (2020-01-14): fix "KeyError: 'ptime'"
This should fix #21 and #30, maybe others
closed 2020-01-14 by amsbr