amsbr/OctoPrint-Stats #22: High CPU usage after update to 2.0. Octoprint unresponsive

pattonwebz, 2019-09-15:
Hey,

I have been using this plugin since for about a year now and it worked great with no issues till the latest update. During the update it took a long time and there is the chance that it failed part way through or timed out.

Ever since the update certain events cause CPU to run 100% locking up the thread and making the interface unresponsive until the plugin finishes it's task. This takes about 3-5 minutes to become responsive again. When this happens during printing it is a visible slowdown (it literally stops between moves), I've just stopped the print anytime this happens so I do not know if it recoverable if it happens during printing.

I have installed `2.0.1pre` but the high cpu load persists.

I checked my system and I have `pandas` and `tinydb` installed. `Numpy` is also installed although I believe I installed that before for a different purpose.

amsbr, 2019-09-16:
Hi @pattonwebz! 

Is it possible to share the stat database file octoprint_stats.json stored in Octoprint logs folder?

pattonwebz, 2019-09-18:
Hey @amsbr sorry for delay in replying here. I am attaching the db file here however after `cat octoprint_stats.db.bak` I can see around all the weird characters that the db contains things about heating failures and some other seemingly error related messages.

Makes me think my database has somehow become corrupted and that would explain the very long processing times and CPU load. I do have a companion `.json` could I perhaps rebuild my database from that file?
[octoprint_stats.db.zip](https://github.com/amsbr/OctoPrint-Stats/files/3626983/octoprint_stats.db.zip)

amsbr, 2019-09-18:
Hi @pattonwebz!

db file used in version 1 of the plugin, it converted to JSON in this version. I will analyze the file in my instance to identify the problem. I will inform you soon.

Do you use a pi3 correct?

pattonwebz, 2019-09-18:
Thank you for your assistance :)

Yes I use a Pi3

gretel, 2019-10-10:
i've also run into this and took me quite a while to figure...
https://github.com/foosel/OctoPrint/issues/3160#issuecomment-540689273

amsbr, 2019-10-11:
Hi,

I go to fix this issue in some days, I'm testing the new version

amsbr, 2019-10-11:
Hi,

Please try the new release, dedicated to reducing CPU usage, and to fix event registration during print. 

[2.0.2](https://github.com/amsbr/OctoPrint-Stats/releases/tag/2.0.2)