amsbr/OctoPrint-Stats #24: v2.0.0 can't install pandas

Ghostbomb, 2019-09-28:
I wanted to install the new update yesterday and pandas took an eternity to install. It never installed cause I rebooted the pi. I feel like this should install in less than 5 min not 2 hours.

From "plugin_softwareupdate_console.log":
`2019-09-27 18:29:43,750 > Installing collected packages: tinydb, pandas, OctoPrint-Stats
2019-09-27 18:29:43,751 > Running setup.py install for pandas: started
2019-09-27 18:30:50,839 > Running setup.py install for pandas: still running...
2019-09-27 18:32:53,006 > Running setup.py install for pandas: still running...
2019-09-27 18:33:59,084 > Running setup.py install for pandas: still running...
2019-09-27 18:46:13,134 > Running setup.py install for pandas: still running...
2019-09-27 19:05:09,360 > Running setup.py install for pandas: still running...
2019-09-27 19:06:34,869 > Running setup.py install for pandas: still running...
2019-09-27 19:07:42,054 > Running setup.py install for pandas: still running...
2019-09-27 19:11:24,322 > Running setup.py install for pandas: still running...
2019-09-27 19:12:31,632 > Running setup.py install for pandas: still running...
2019-09-27 19:14:11,786 > Running setup.py install for pandas: still running...
2019-09-27 19:15:12,128 > Running setup.py install for pandas: still running...
2019-09-27 19:17:57,788 > Running setup.py install for pandas: still running...
2019-09-27 19:19:32,975 > Running setup.py install for pandas: still running...
2019-09-27 19:33:20,096 > Running setup.py install for pandas: still running...
2019-09-27 19:35:56,270 > Running setup.py install for pandas: still running...
2019-09-27 19:38:03,467 > Running setup.py install for pandas: still running...
2019-09-27 19:43:31,996 > Running setup.py install for pandas: still running...
2019-09-27 19:45:41,266 > Running setup.py install for pandas: still running...
2019-09-27 19:46:50,339 > Running setup.py install for pandas: still running...
2019-09-27 19:48:23,784 > Running setup.py install for pandas: still running...
2019-09-27 19:49:35,807 > Running setup.py install for pandas: still running...
2019-09-27 19:51:37,155 > Running setup.py install for pandas: still running...
2019-09-27 19:54:18,329 > Running setup.py install for pandas: still running...
2019-09-27 19:56:00,857 > Running setup.py install for pandas: still running...
2019-09-27 19:58:44,056 > Running setup.py install for pandas: still running...
2019-09-27 20:00:56,511 > Running setup.py install for pandas: still running...
2019-09-27 20:03:43,706 > Running setup.py install for pandas: still running...
2019-09-27 20:04:50,857 > Running setup.py install for pandas: still running...`

ThorstenWirth, 2019-10-02:
You'll have to ssh  into your Octopi and then run
~/oprint/bin/pip --no-cache-dir install pandas
This will take some time.

Just to make sure everything goes smoothly install Byobu first and run
byobu-enable
- this will keep the terminal session alive, even when the ssh-connection is interrupted.

Ghostbomb, 2019-10-08:
yep fixed it. Took forever tho

closed 2019-10-08 by Ghostbomb