amsbr/OctoPrint-Stats #25: I got something wrong related to theese pandas too! Could someone help me what to do?

lucaskoeler, 2019-10-03:
Installing plugin from uploaded archive...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install /tmp/tmpwEUtl4.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpwEUtl4.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.0) (1.3.11)
Collecting tinydb (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/dc/90/ced9f2a57513eb75dce70760e55967ceedbf2dc158b6a66f1bf87ac8fab0/tinydb-3.14.1-py2.py3-none-any.whl
Collecting pandas (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-xWoyaY/pandas/setup.py", line 746, in <module>
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 724, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 846, in resolve
dist = best[req.key] = env.best_match(req, ws, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1118, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1130, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 791, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-unT_Q5/numpy-1.17.2/setup.py", line 31, in <module>
RuntimeError: Python version >= 3.5 required.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-xWoyaY/pandas/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

condatek, 2019-10-10:
Same issue when try installing it in Octoprint 1.3.11 (manually installed in a fresh Raspbian on a RPi 2)

condatek, 2019-10-11:
Hi, yes, I first tried running commands of #24 but same results. The problem was numpy has to be installed before the command. Here's how I fixed it: 

`cd ~/OctoPrint/venv/bin/
source ./venv/bin/activate
./venv/bin/pip --no-cache-dir install numpy
./venv/bin/pip --no-cache-dir install pandas`

amsbr, 2019-10-11:
Very good!

I'm looking for a way to improve the installation process because of numpy dependency

oblivionratula, 2019-10-13:
Thanks for this guidance, condatek, Both numpy and pandas took a LONG time to build on a pi, but the plugin installs now.