amsbr/OctoPrint-Stats #26: Please bring back v1

purist69, 2019-10-05:
Sorry, but your v2 is terrible.  It looks pretty, when it finally loads, but it's extremely slow, takes forever to paint the page.  the default is past year!?!?!?  That's too much data to load!

I run Octoprint on a very fast RK3399 chipset, there is no reason for your plugin to be so slow on such a fast processor and in 64bit mode.

The install is terrible, if it works at all....Armbian doesn't have a large enough swap partition and pandas fails to install.  The install works fine on DietPi AFTER getting numpy and pandas installed, but pandas takes FOREVER to install.  I spent over a day working on this to get it working because I loved v1 soo much.

Other than looking prettier, it doesn't provide that much more functionality than v1 and it way too slow.  I don't think you will have a large install base until you fix the install issues and performance issues.

amsbr, 2019-10-10:
Hi @purist69 

Thanks for your feedback!

The migration to pandas is exactly so you can support new features and statistics. This was not cool with SQLite.

I am still evaluating how to improve the installation process, I haven't found a way yet.

Already regarding performance I am working to bring improvements in this direction in the coming days.

amsbr, 2019-10-11:
@purist69 

Try the new release, dedicated to reducing CPU usage, and to fix event registration during print. 

[2.0.2](https://github.com/amsbr/OctoPrint-Stats/releases/tag/2.0.2)

Other features will be released soon.

purist69, 2019-10-11:
I'm using this on a Rock64 right now in 64bit mode, CPU still clocks at 100% while retrieving data.  My concern is this is in the same process as the print job.  I don't want any failed prints from excessive CPU useage.

amsbr, 2019-10-11:
I understand, the version 2.0.0 have a big bug, in this version the events run refresh data function in all moments (including during print job), the version 2.0.2, I'm removed all events during job process (it's not necessary, not now).

The insert into database not use much of CPU, but refresh data function use much CPU (this process now only in stats tab or side bar during filter function), I'm looking for a away to optimize this too using pandas library.

The primary problem was solved, restrict insert events into database and run refresh data function only on sidebar show or tab stats, and default filter set to Current Month.

purist69, 2019-10-11:
If you have a way to cache the data and just refresh the delta vs. A full data pull every time, that would help. Or possibly a DB view?

amsbr, 2019-10-11:
Yes, that's exactly right, besides working with caching, reducing file system access