amsbr/OctoPrint-Stats #27: Can't Updated - Log Attached

icancwhatusay, 2019-10-07:
Octopi didn't give any specific reason as to  why it failed, just fails.

This is the only warning message I get:

[image]
[octoprint.log](https://github.com/amsbr/OctoPrint-Stats/files/3697931/octoprint.log)

amsbr, 2019-10-10:
Hi @icancwhatusay 

Please try run the commands of issue #24

icancwhatusay, 2019-10-10:
I'll give it a go but without a way to get the IP address off the pi remotely, this will take some time. (be great if there were a way to access the terminal from Octopi)

icancwhatusay, 2019-10-25:
[plugin_softwareupdate_console(2).log](https://github.com/amsbr/OctoPrint-Stats/files/3773624/plugin_softwareupdate_console.2.log)

icancwhatusay, 2019-10-25:
OK so I finally was able to get my ip address from my IT guy. 

1) before I started I did the following:

sudo apt-get update
sudo apt-get upgrade

and let that run. 

Then I ran the code you suggested:
~/oprint/bin/pip --no-cache-dir install pandas

and then I get this error
[image]

condatek, 2019-10-25:
You have to try this before that commnland:
'./pip --no-cache-dir install numpy'

icancwhatusay, 2019-10-28:
_In case this wasn't included before_
```
------------------------------------------------------------------------------
OctoPrint version : 1.3.12
OctoPi version    : 0.15.1
------------------------------------------------------------------------------
```

OK Here are my steps: 

Install PuTTY
SSH into pi using and the ip address from the pi - obtained by logging into my network admin portal

On the pi:
```
sudo apt-get update
sudo apt-get install python-pip #says it installed 19.3.1 but when I type sudo pip -V, it came back with 9.0.1   SO....

sudo pip install --upgrade pip
```
[image]

```
sudo apt-get upgrade
./pip --no-cache-dir install numpy #installs fine according to output logs (this screen shot is just a double check, thus the 'requirement already satisfied"
```

[image]

_OK Now on to the pandas install_
```
sudo ~/oprint/bin/pip --no-cache-dir install pandas

```
_ERROR: message_
Same as before "You are using pip version 9.0.3, however version 19.3.1 is available."

[image]

And just to verify that I still have 19.3.1

[image]

Further, (hopefully this is related and not a separate issue) I can no longer access octopi.local I get a 500 error. "Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

Hope this helps.

icancwhatusay, 2019-10-28:
> Updating, please wait.
+++++++++++++++++++++++++++++++++++
Now updating Printer Stats to 2.0.2
+++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/2.0.2.zip --no-cache-dir
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/2.0.2.zip
Downloading https://github.com/amsbr/OctoPrint-Stats/archive/2.0.2.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.2)
Collecting tinydb (from OctoPrint-Stats==2.0.2)
Downloading https://files.pythonhosted.org/packages/9b/83/2d46115b89640e9b85b94df47216547396e94125245dd3ade186036ce976/tinydb-3.15.1-py2.py3-none-any.whl
Collecting pandas (from OctoPrint-Stats==2.0.2)
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-build-XdzkIe/pandas/setup.py", line 746, in <module>
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 128, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 123, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 504, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 846, in resolve
dist = best[req.key] = env.best_match(req, ws, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1118, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1130, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 571, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 673, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 699, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 884, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1152, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1138, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-ubcmN9/numpy-1.17.3/setup.py", line 31, in <module>
RuntimeError: Python version >= 3.5 required.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-XdzkIe/pandas/
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

amsbr, 2019-10-30:
Hi,

Try run first:

pip install numpy==1.10.4  (long time for compile/install in arm boards)

icancwhatusay, 2019-10-30:
[image]

icancwhatusay, 2019-10-30:
Got it with adding sudo before the code.

[image]

But then I get this when I try the next steps:

[image]

closed 2021-05-25 by icancwhatusay