amsbr/OctoPrint-Stats #28: Installing plugin fails on fresh Octopi 0.16.0

fleibede, 2019-10-08:
Model: Raspberry Pi 3 Model B Rev 1.2
Applied OctoPi 0.16.0 image. 
Updated to 1.3.11, installed plugins themeify, psucontrol and finally this plugin.

RuntimeError: Python version >= 3.5 required?

Should this not work out of the box on this setup? See pluginmanager log attached.
[plugin_pluginmanager_console.log](https://github.com/amsbr/OctoPrint-Stats/files/3704913/plugin_pluginmanager_console.log)

fleibede, 2019-10-08:
Same issue as https://github.com/amsbr/OctoPrint-Stats/issues/25

NUsulator, 2019-10-13:
i had to launch in SSH:
~/oprint/bin/pip --no-cache-dir install numpy
~/oprint/bin/pip --no-cache-dir install pandas

then i install the plugin

padouk73, 2020-02-07:
after install numpy abd install pandas, reboot must be done, than go to plugin manager and install statistics plugin.
Works !

redfox05, 2020-03-06:
+1 issue is still happening.
Octoprint Version 1.4.0 fresh install, on Raspberry Pi 3.

This needs a proper fix, manually installing packages on the command line is not good for a system like OctoPrint, as that can bring unexpected problems later on.

Example of why it's bad: Some threads recommend manually installing Python 3.5 or above, and this is BAD, because a lot of the OctoPrint system is not compatible with 3.x

So, case in point, a proper fix is needed.