amsbr/OctoPrint-Stats #32: Error Install Plugin last release...

bgiovanny, 2019-12-14:
Installing plugin "Printer Statistics" from https://github.com/amsbr/OctoPrint-Stats/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Downloading https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.2) (1.3.12)
Collecting tinydb
Downloading https://files.pythonhosted.org/packages/9b/83/2d46115b89640e9b85b94df47216547396e94125245dd3ade186036ce976/tinydb-3.15.1-py2.py3-none-any.whl
Collecting pandas
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-n5bIQc/pandas/setup.py'"'"'; file='"'"'/tmp/pip-install-n5bIQc/pandas/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(file);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-n5bIQc/pandas/pip-egg-info
cwd: /tmp/pip-install-n5bIQc/pandas/
Complete output (51 lines):
Traceback (most recent call last):
File "", line 1, in
File "/tmp/pip-install-n5bIQc/pandas/setup.py", line 746, in
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/init.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/init.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 720, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/init.py", line 782, in resolve
replace_conflicting=replace_conflicting
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/init.py", line 1065, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/init.py", line 1077, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 787, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in exit
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in exit
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-Gpmr7K/numpy-1.18.0rc1/setup.py", line 31, in
RuntimeError: Python version >= 3.5 required.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

ZhenyaKa, 2019-12-24:
+1 can't update to 2.0
Hangs while installing pandas.

Orange Pi Lite, Orange Pi Zero with Armbian

EpicSalvation, 2020-01-02:
This error is because the current versions of Numpy and Pandas don't suppport Python 2.7.  I had to SSH into the Pi and run the following commands to install the latest versions that do support Python 2.7.  This takes a while so installing Byobu keeps the session alive even if your SSH client disconnects.  Note that order matters here: Numpy needs to be installed before Pandas.

_sudo apt-get install byobu
byobu-enable
~/oprint/bin/pip --no-cache-dir install numpy==1.16
~/oprint/bin/pip --no-cache-dir install pandas==0.24.2_

ZhenyaKa, 2020-01-03:
run_setup(setup_script, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 258, in run_setup
        raise
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 198, in setup_context
        yield
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 169, in save_modules
        saved_exc.resume()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 144, in resume
        six.reraise(type, exc, self._tb)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 157, in save_modules
        yield saved
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 198, in setup_context
        yield
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 255, in run_setup
        DirectorySandbox(setup_dir).run(runner)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 285, in run
        return func()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 253, in runner
        _execfile(setup_script, ns)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 47, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-rRo3GW/numpy-1.18.0/setup.py", line 31, in <module>

    RuntimeError: Python version >= 3.5 required.

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-zfoBJa/pandas/

ZhenyaKa, 2020-01-03:
numpy installed okay

evannadeau, 2020-02-08:
I had an issue installing the latest version as well, and followed the above steps, and it is now up and running.  Thanks.

drehzahl, 2020-02-08:
🙄 i am not a Crack, i am a User. I wait for a normal installation.

radiumhost, 2020-03-06:
+1 issue is still happening.
Octoprint Version 1.4.0 fresh install, on Raspberry Pi 3.

redfox05, 2020-03-06:
+1 issue is still happening.
Octoprint Version 1.4.0 fresh install, on Raspberry Pi 3.