amsbr/OctoPrint-Stats #33: Stats don't load on new install, installed after fix 31

SoLongSidekick, 2020-01-05:
Just installed today after having to manually regress Numpy to version 1.16 and Pandas to version 0.24.2 because the current versions do not support Python 2.7. I noticed that someone else registered an issue for this same problem, and a fix was mentioned but that fix was implemented well before today when I installed. 

To be clear, the stats never load and the loading animation just plays continually.

drehzahl, 2020-01-14:
Same here Octoprint 0.17 fresh installed today 
RuntimeError: Python version >= 3.5 required.

goldjunge91, 2020-02-25:
same her can not install
`Downloading pandas-0.24.2.tar.gz (11.8 MB)
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-OHAe7C/pandas/setup.py'"'"'; __file__='"'"'/tmp/pip-install-OHAe7C/pandas/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-OHAe7C/pandas/pip-egg-info
cwd: /tmp/pip-install-OHAe7C/pandas/
Complete output (51 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-OHAe7C/pandas/setup.py", line 746, in <module>
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 720, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 782, in resolve
replace_conflicting=replace_conflicting
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1065, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1077, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 787, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-SY30im/numpy-1.18.1/setup.py", line 32, in <module>
RuntimeError: Python version >= 3.5 required.
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

Hobb3s, 2020-03-03:
same here with the spilling animation and no stats loading.