amsbr/OctoPrint-Stats #36: Support for OctoPrint 1.4.0

SargonofAssyria, 2020-03-08:
Please support for python 3.7 via the new plugin property __plugin_pythoncompat__

benjamin-salchow, 2020-03-15:
I had the same issue. I created a Pull Request: https://github.com/amsbr/OctoPrint-Stats/pull/37 with informations to run it with python 3.7.

But keep in mind, that you need to install a Raspbian package:

```
apt install libatlas-base-dev
```

You can install it from my repository as zip: https://github.com/benjamin-salchow/OctoPrint-Stats/archive/master.zip

SargonofAssyria, 2020-03-24:
Tnx, that did it. It's working now.
I put ___plugin_pythoncompat__ = ">=2.7,<4"_ in several other plugins as well.
Pitty that @amsbr does not respond.

closed 2020-03-24 by SargonofAssyria

gvmturl, 2020-04-02:
> Tnx, that did it. It's working now.
> I put _**plugin_pythoncompat** = ">=2.7,<4"_ in several other plugins as well.
> Pitty that @amsbr does not respond.

How to do that bro? Still having problems here....

> I had the same issue. I created a Pull Request: #37 with informations to run it with python 3.7.
>
> But keep in mind, that you need to install a Raspbian package:
>
> ```
> apt install libatlas-base-dev [...]

Not worked!

Anyhelp for that?

benjamin-salchow, 2020-04-03:
Just run:
```apt install libatlas-base-dev```

and afterwards remove the plug-in and install it from: https://github.com/benjamin-salchow/OctoPrint-Stats/archive/master.zip 

This contains the `plugin_pythoncompat` and should work. Works on my Python 3 set-up like a charm.

@amsbr could you please take a look at my pull request?

gvmturl, 2020-04-03:
> Just run:
> `apt install libatlas-base-dev`
>
> and afterwards remove the plug-in and install it from: https://github.com/benjamin-salchow/OctoPrint-Stats/archive/master.zip
>
> This contains the [...]

Thanks for replying!
am i doing something wrong?
1# Access Raspberry by Putty
2# sudo apt install libatlas-base-dev
3# sudo reboot
4# Install plugin using Plugin Manager pasting the link:
https://github.com/benjamin-salchow/OctoPrint-Stats/archive/master.zip

is it correct?
Maybe some broken data made by fails installation making me this troble?

Thank you!!!

benjamin-salchow, 2020-04-03:
You're welcome. The steps are correct that you wrote. 

By looking at the log it looks like an issue with your installation. Because the log says that you're using "/home/pi/oprint/bin/python2" - but your installation of numpy and panda is for Python version >= 3.5. Try to resetup/reinstall the virtual python environment with Python 3.

Then it should work and in the log should be no "pyhton2" anymore ;-)

gvmturl, 2020-04-03:
> reinstall the virtual python environment with Python 3

Hello Benjamin! I m sorry, could you explain me how to?

Thank your for time bro.

gvmturl, 2020-04-05:
Any help please?
This plugin is so important for what i need.

benjamin-salchow, 2020-04-05:
Hello,

follow the instructions here: https://github.com/OctoPrint/OctoPrint#installation

Do a reinstall of OctoPrint and use the following commands instead of the shown version in this link I shared:
```sh
virtualenv --python=python3.7 venv
# instead of: virtualenv --python=python2.7 venv
```

Then you move on with the installation and afterwards you add all the plugins to it.

If you do it this way, all will be correct. Use `virtualenv` and not the Python Installation that you wrote 2 days ago. In your system there will be now a mix of Python2 from the system and Python3 that you installed manually on it. If it does not work I would recommend to reinstall the Pi complete. The manual installation could damage system-packages and the repair of it would take even longer than a new installation from scratch.

Hope that helps ;-).

gvmturl, 2020-04-06:
Brother, thank you again.
I m really sorry. I m starting everything again.
To start must i download the image and install it with Raspberry Imager, then change values to connect on wifi...

Login to Putty as pi then follow those steps?
Checkout OctoPrint: git clone https://github.com/OctoPrint/OctoPrint.git
Change into the OctoPrint folder: cd OctoPrint
Create a user-owned virtual virtualenv --python=python3.7 venv
Install OctoPrint into that virtual environment: ./venv/bin/pip install 

Not sure if it is... i m getting errors. tried something else

sudo apt update
source venv/bin/activate
pip install pip --upgrade
pip install octoprint

but still have some errors on octoprint installation... Its is installed... but still getting the same problem to install plugin...

Could you try to show me what i m doing wrong please?  I m sorry and thank you again.

benjamin-salchow, 2020-04-06:
Hello,
try an installation with the standard Raspbian image. If this also not work I'm really not sure what the problem is. Looks like a damaged pyhton2 and python3 setup to me. I'm not using the Octoprint Image, so can't test that.

gvmturl, 2020-04-07:
> Hello,
> try an installation with the standard Raspbian image. If this also not work I'm really not sure what the problem is. Looks like a damaged pyhton2 and python3 setup to me. I'm not using the Octoprint Image, so can't test that.

Hello man, thank you. Solved the problem by following these steps:
`pip install --upgrade pip 
git clone https://github.com/foosel/OctoPrint.git
virtualenv --python=python3.7 venv 
cd OctoPrint  
source venv/bin/activate

sudo apt-get install python3-setuptools python3-pyqt5 python3-pip
git checkout devel 
python setup.py clean 
pip install -e .[develop,plugins] 
octoprint serve 

Clear oprint folder that is on pi@octopi then copied all content of venv folder, so octopi image will start on python 3 env
`
Now i having problems with some plugins that are not compatible with python 3, found some on your repository, thank you. But still have some that i d like to use. How can i make then compatible with this version of python? Tried using 2to3 library but didn't work. Are you going to translating more plugins? Some repo seems to be died T_T .

Thank you bro.

benjamin-salchow, 2020-04-07:
I'm glad that I could help you. Yes, many plug-ins are not maintained any more. All of my "merge" requests are not integrated yet. I did not use the 2to3 library, I just changed the source code. Take a look at my changes I did in the plug-ins. In many it is just that line : `__plugin_pythoncompat__ = ">=2.7,<4"`

E.g. here: https://github.com/amsbr/OctoPrint-Stats/pull/37/commits/81878c03f76020d84df7b39e77f58bc828a19948

See the deltas of my changes and try it for your plug-ins that don't have any python 3 compatibly. Then you have to test it and see in the logs, if there are other errors. That's a pragmatic way to do it - but the fastest. The Python 2 to 3 was started way to late in OctoPrint.. The EOL of Python 2 is well known since years.

gvmturl, 2020-04-12:
> I'm glad that I could help you. Yes, many plug-ins are not maintained any more. All of my "merge" requests are not integrated yet. I did not use the 2to3 library, I just changed the source code. Take a look at my changes I did in the plug-ins. In many it is just that line : `__plugin_pythoncompat__ = ">=2.7,<4"`
> 
> E.g. here: [81878c0](https://github.com/amsbr/OctoPrint-Stats/commit/81878c03f76020d84df7b39e77f58bc828a19948)
> 
> See the deltas of my changes and try it for your plug-ins that don't have any python 3 compatibly. Then you have to test it and see in the logs, if there are other errors. That's a pragmatic way to do it - but the fastest. The Python 2 to 3 was started way to late in OctoPrint.. The EOL of Python 2 is well known since years.

Hello Benjamin, thank you. worked on the following plugins
OctoPrint-FullScreen
OctoPrint-Autoselect
OctoPrint-Autoscroll
OctoPrint-ExcludeRegionPlugin

How can check the logs?
Where is located the folder with the installed plugins?

Maybe could you take a look on those plugins?
https://github.com/fabianonline/OctoPrint-Telegram
https://github.com/devildant/OctoPrint-ShutdownPrinter

Thank you again!

foosel, 2020-04-15:
I just came across this why following up on OctoPrint/plugins.octoprint.org#434... To clarify, OctoPrint 1.4.0 does not require Python 3 to function. A dependency of this plugin however does apparently (and it should either be pinned to an older version that still supports Python 2 or this plugin needs to be migrated to Python 3 only and marked accordingly on the repository).

Plugins having Python 3 only dependencies has nothing to do with OctoPrint 1.4.0. I would appreciate it if people would stop spreading this non sense that OctoPrint will only work under Python 3 now, due to issues with third party plugins that have failed to pin their dependencies to versions known to work properly.

SargonofAssyria, 2020-04-15:
It might work, but read this: https://www.python.org/doc/sunset-python-2/

foosel, 2020-04-15:
You honestly think I don't know that? I spent the past year doing nothing but prepping OctoPrint 1.4.0 to be Python 3 *and* 2 ready, so that it would be prepared for both people knowing that and people ignoring that.

benjamin-salchow, 2020-05-01:
Hello @foosel ,
first thank you very much for your great work on OctoPrint. It's really a great software for my 3D printer and I'm glad to use it for all my prints. In addition, a big thank you for maintaining OctoPrint so good.
You're right, my pull request was not tested with Python 2 and regarding the dependencies it won't work anymore with Python 2. 
I run OctoPrint complete with Python 3 so I created many pull request to some plug-ins that didn't utilize Python 3 Support. In most cases this is easily done. In some cases there are dependencies used that require now Python 3, like here with numpy.
This issue is not good, because as you mentioned OctoPrint 1.4.0 runs on Python 2 as Python 3, so the issue is, that this plug-in wasn't Python 3 compatible. I hope the maintainer of this plug-in will shortly take a look at it and change it. But many plug-ins I committed a pull request months ago did not answer or support it in any way but I hope that this could be solved soon. I don't know what the best practice is to utilize Python 2 and Python 3 support, if the dependency requires a dedicated Python version.

foosel, 2020-05-04:
> I don't know what the best practice is to utilize Python 2 and Python 3 support, if the dependency requires a dedicated Python version.

Version pinning & environment markers/dynamic environment detection. I do that in OctoPrint too, exactly due to the mentioned problem:

https://github.com/OctoPrint/OctoPrint/blob/ef3c8bb6c24f2f94b6c2f486e96c7fe098f06dcd/setup.py#L114-L129

Or alternatively marking the plugin as Python 3 only here and on the repository.

> But many plug-ins I committed a pull request months ago did not answer or support it in any way

I fear the Python 3 switch will reveal quite a number of plugins that are now unmaintained/abandoned by their authors. If you come across stuff like that please use the link to report abandoned plugins on the plugin repository, since that at least will show people the current state of affairs and maybe help in recruiting a new maintainer.