amsbr/OctoPrint-Stats #40: [Compatibility] Only supports Python 3 in Octoprint 1.4.0

VonOgre, 2020-04-15:
I tried to install the plugin on an instance of Octoprint 1.4.0 that is still running on Python 2, since I was doing in-place upgrades and didn't realize there's a migration path that needs to be performed manually to get to Python 3. The latest release lists a dependency on python >=2.7, < 3, but fails because numpy needs Python 3.7.

Linking issue in Octoprint's plugins tracker: https://github.com/OctoPrint/plugins.octoprint.org/issues/434

Here is the install log for the plugin:
[plugin_pluginmanager_console.log](https://github.com/amsbr/OctoPrint-Stats/files/4482122/plugin_pluginmanager_console.log)

IIAIronWolf, 2020-04-23:
This is a pretty important plugin for my printer. Hopefully the author responds soon.

purist69, 2020-04-29:
I believe I got it working with 1.4.0, this is what I did:

First, go to the plugin manager and attempt a manual install, I don't know if there are things this installed, so just do it:
install https://github.com/amsbr/OctoPrint-Stats/archive/master.zip

Second, Go to the plugin manager and see the install location of your pip mine was:
/home/pi/oprint/bin aka ~/oprint/bin

Third, install pandas:
~/oprint/bin/pip install pandas - this will take a looooooong time

Fourth, install numpy
~/oprint/bin/pip install numpy-1.16.6

Fifth, go back to the plugin manager and do a manual install again:
https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
this will get an error at the end, but after a restart of octoprint the plugin should start working.  I did an upload of a file and the stats appeared.