amsbr/OctoPrint-Stats #48: Future of this plugin

AlexVerrico, 2020-09-25:
Hi @amsbr 
Are you able to continue maintaining this plugin (at least merging Pull Requests like #37 )?
If you are unable to commit to at least merging PRs, are you open to transferring the repo to a new owner so that some of the issues can be fixed and the plugin can continue to be used?
i'm happy to discuss me taking over this repo (which would be best), otherwise if I don't hear back from you within 14 days then I am going to mirror the repo so that i can begin fixing some of the issues, and eventually update the listing in the plugin repository to point to my repo so that people can continue to use this plugin.
Cheers,
Alex.

AlexVerrico, 2020-10-01:
Hi @amsbr 
I would really prefer to take over this repo rather than mirroring it, it would be great if you could let me/the community know what the go is here.
Cheers,
Alex.

AlexVerrico, 2020-10-10:
As there has been no reply, I have mirrored this repo, and it is now available here: https://github.com/AlexVerrico/octoprint-stats
I will be working on fixing bugs and adding requested features as I can, PRs are always welcome though 😉

amsbr, 2020-10-18:
> Hi @amsbr
> Are you able to continue maintaining this plugin (at least merging Pull Requests like #37 )?
> If you are unable to commit to at least merging PRs, are you open to transferring the repo to a new owner so that some of the issues can be fixed and the plugin can continue to be used?
> i'm happy to discuss me taking over this repo (which would be best as it would transfer existing PRs and issues), otherwise if I don't hear back from you within 14 days then I am going to mirror the repo so that i can begin fixing some of the issues, and eventually update the listing in the plugin repository to point to my repo so that people can continue to use this plugin.
> Cheers,
> Alex.

Hi @AlexVerrico,

As I will not be able to maintain the plugin, I will be happy that someone else will be able to maintain it.

amsbr, 2020-10-18:
> As there has been no reply, I have mirrored this repo, and it is now available here: https://github.com/AlexVerrico/octoprint-stats
> I will be working on fixing bugs and adding requested features as I can, PRs are always welcome though 😉

I hope to be able to contribute in the future. Thank you!

AlexVerrico, 2020-10-19:
PRs are always welcome if you want to.